OpenTimer/OpenTimer #1: Failed to compile.

ohmpossum, 2017-11-08:
I was attempting to follow the udemy free tools install course "vsd a complete guide to install open source eda tools" chapter 5 about OpenTimer. I was unable to compile OpenTimer-1.0.5 I haven't written any C++ since perhaps 1995 but was able to resolve the issue. In the file: demangle.cc I changed two lines:
line 170
    if (*str == '\0') {
//    if (str == '\0') { // maybe fixed GFA 11/7/17 with above line added *

and this around line 226: 

  if (*state->mangled_cur == '\0') {
//  if (state->mangled_cur == '\0') { //original replaced by above GFA 11/07/17

After doing that and figuring out after compile, I need to run a command "sudo ldconfig" this now compiles and runs under Ubuntu 17.10. Perhaps this or what ever was intended could be included in the source code for this project.

Thanks,
Gordon Aplin

Tsung-Wei, 2017-11-08:
What is your gcc version? It's better to have GCC 6.0

ohmpossum, 2017-11-08:
Thanks for the quick reply. 

It is gcc 7.2.0.

Tsung-Wei, 2017-11-08:
Yes. GCC 7 will be fine.