OpenTimer/OpenTimer #2: cmake issue with compiling OpenTimer 2.0

eeegar, 2018-07-25:
Hello Tsung-Wei

I tried to compile the 2.0 code but hitting the issues:
If you have a chance, please take a look. I am not an experienced C developer, just compile when I need to. So if it rings any bell, please let me know. I am trying to compile it for my PC using Cygwin environment.

Thanks in advance!

Igor  

$ cmake ../
-- CMAKE_ROOT: /usr/share/cmake-3.6.2
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++.exe
-- Check for working CXX compiler: /usr/bin/c++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type:
-- CMAKE_SYSTEM_NAME: CYGWIN
-- CMAKE_CXX_COMPILER: g++
-- CMAKE_CXX_FLAGS:  -Wall -O2
-- CMAKE_INSTALL_PREFIX: /usr/local
-- OT_VERSION: 2.0.0 (alpha)
-- OT_HOME: /cygdrive/c/Users/IGOR/Theseus/CAD/OpenTimer/OpenTimer-master
-- OT_BENCHMARK_DIR: /cygdrive/c/Users/IGOR/Theseus/CAD/OpenTimer/OpenTimer-master/benchmark
-- OT_UNITTEST_DIR: /cygdrive/c/Users/IGOR/Theseus/CAD/OpenTimer/OpenTimer-master/unittest
-- OT_LICENSE: /cygdrive/c/Users/IGOR/Theseus/CAD/OpenTimer/OpenTimer-master/LICENSE
-- CMAKE_ARCHIVE_OUTPUT_DIRECTORY: /cygdrive/c/Users/IGOR/Theseus/CAD/OpenTimer/OpenTimer-master/lib
-- OT_CPP: ot/unit/unit.cpp;ot/shell/misc.cpp;ot/shell/obselete.cpp;ot/shell/builder.cpp;ot/shell/action.cpp;ot/shell/shell.cpp;ot/shell/dump.cpp;ot/timer/arc.cpp;ot/timer/celllib.cpp;ot/timer/test.cpp;ot/timer/timer.cpp;ot/timer/clock.cpp;ot/timer/sdc.cpp;ot/timer/endpoint.cpp;ot/timer/net.cpp;ot/timer/sfxt.cpp;ot/timer/unit.cpp;ot/timer/path.cpp;ot/timer/spef.cpp;ot/timer/cppr.cpp;ot/timer/verilog.cpp;ot/timer/gate.cpp;ot/timer/dump.cpp;ot/timer/pin.cpp;ot/liberty/celllib.cpp;ot/liberty/cell.cpp;ot/liberty/cellpin.cpp;ot/liberty/lut.cpp;ot/liberty/timing.cpp;ot/verilog/verilog.cpp;ot/sdc/tokenize.cpp;ot/sdc/object.cpp;ot/sdc/sdc.cpp;ot/tau/tau15.cpp;ot/utility/os.cpp;ot/utility/tokenizer.cpp;ot/spef/spef.cpp
-- OT_HPP: ot/unit/unit.hpp;ot/policy.hpp;ot/shell/shell.hpp;ot/shell/prompt.hpp;ot/traits.hpp;ot/timer/cppr.hpp;ot/timer/gate.hpp;ot/timer/pin.hpp;ot/timer/sfxt.hpp;ot/timer/path.hpp;ot/timer/timer.hpp;ot/timer/test.hpp;ot/timer/clock.hpp;ot/timer/endpoint.hpp;ot/timer/net.hpp;ot/timer/arc.hpp;ot/config.hpp;ot/headerdef.hpp;ot/liberty/lut.hpp;ot/liberty/timing.hpp;ot/liberty/cellpin.hpp;ot/liberty/cell.hpp;ot/liberty/celllib.hpp;ot/verilog/verilog.hpp;ot/static/logger.hpp;ot/static/thread.hpp;ot/sdc/object.hpp;ot/sdc/sdc.hpp;ot/sdc/mask.hpp;ot/sdc/tokenize.hpp;ot/taskflow/taskflow.hpp;ot/tau/tau15.hpp;ot/exception.hpp;ot/utility/logger.hpp;ot/utility/scope_guard.hpp;ot/utility/utility.hpp;ot/utility/index.hpp;ot/utility/lambda.hpp;ot/utility/tokenizer.hpp;ot/utility/CLI11.hpp;ot/utility/singleton.hpp;ot/utility/os.hpp;ot/utility/iterator.hpp;ot/spef/spef.hpp
-- PYTHON_EXECUTABLE: /usr/bin/python
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE
-- Building executables ...
-- OT_EXE_LINKER_FLAGS: OpenTimer;Threads::Threads;stdc++fs
-- TAU15 executable: /cygdrive/c/Users/IGOR/Theseus/CAD/OpenTimer/OpenTimer-master/bin/ot-tau15
-- Utility executable: /cygdrive/c/Users/IGOR/Theseus/CAD/OpenTimer/OpenTimer-master/bin/ot-utility
-- OpenTimer shell: /cygdrive/c/Users/IGOR/Theseus/CAD/OpenTimer/OpenTimer-master/bin/ot-shell
-- Building unit tests ...
-- Building Shell integration tests ...
-- Building TAU15 integration tests ...
-- Configuring done
CMake Error in CMakeLists.txt:
  Target "sdc" requires the language dialect "CXX17" , but CMake does not
  know the compile flags to use to enable it.

CMake Error in CMakeLists.txt:
  Target "unit" requires the language dialect "CXX17" , but CMake does not
  know the compile flags to use to enable it.

CMake Error in CMakeLists.txt:
  Target "ot-tau15" requires the language dialect "CXX17" , but CMake does
  not know the compile flags to use to enable it.

CMake Error in CMakeLists.txt:
  Target "OpenTimer" requires the language dialect "CXX17" , but CMake does
  not know the compile flags to use to enable it.

CMake Error in CMakeLists.txt:
  Target "ot-utility" requires the language dialect "CXX17" , but CMake does
  not know the compile flags to use to enable it.

CMake Error in CMakeLists.txt:
  Target "ot-shell" requires the language dialect "CXX17" , but CMake does
  not know the compile flags to use to enable it.

CMake Error in CMakeLists.txt:
  Target "s27" requires the language dialect "CXX17" , but CMake does not
  know the compile flags to use to enable it.

CMake Error in CMakeLists.txt:
  Target "utility" requires the language dialect "CXX17" , but CMake does not
  know the compile flags to use to enable it.

-- Generating done
-- Build files have been written to: /cygdrive/c/Users/IGOR/Theseus/CAD/OpenTimer/OpenTimer-master/build

twhuang-uiuc, 2018-07-25:
Hi, it turns out your cmake can't support C++17 flag, and I didn't put it right at the project `CMakeLists.txt`. Would you please upgrade your cmake to version **at least 3.9** and `git pull` the master branch to retry it again?

eeegar, 2018-07-27:
Hello Tsung-Wei,

For Cygwin, I can see the latest cmake package available is 3.6.2-1
I also tried to compile the 3.12 release of cmake from source, but it asked for something else.... so peeling the onion here. 

Igor

twhuang-uiuc, 2018-07-29:
Yes. I believe it's due to the C++17 issue. I highly recommend getting a Linux machine (or virtual box) to use.

closed 2018-09-11 by twhuang-uiuc