OpenTimer/OpenTimer #4: Compile error during installation

userghgh, 2018-07-31:
**Describe the bug**

While building OpenTimer on OS X Yosemite, **cmake** runs fine, but **make** cannont find header files string_view, variant, experimental/filesystem, optional:

$ cmake ../
-- CMAKE_ROOT: /Applications/CMake.app/Contents/share/cmake-3.12
-- The C compiler identification is AppleClang 7.0.2.7000181
-- The CXX compiler identification is AppleClang 7.0.2.7000181
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_CXX_COMPILER_VERSION: 7.0.2.7000181
-- CMAKE_BUILD_TYPE: 
-- CMAKE_SYSTEM_NAME: Darwin
-- CMAKE_CXX_COMPILER: g++
-- CMAKE_CXX_FLAGS:  -Wall -O2
-- CMAKE_INSTALL_PREFIX: /usr/local
-- CMAKE_MODULE_PATH: 
-- OT_VERSION: 2.0.0 (alpha)
-- OT_HOME: /Users/blah/vlsi/OpenTimer
-- OT_BENCHMARK_DIR: /Users/blah/vlsi/OpenTimer/benchmark
-- OT_UNITTEST_DIR: /Users/blah/vlsi/OpenTimer/unittest
-- OT_LICENSE: /Users/blah/vlsi/OpenTimer/LICENSE
-- CMAKE_ARCHIVE_OUTPUT_DIRECTORY: /Users/blah/vlsi/OpenTimer/lib
-- PYTHON_EXECUTABLE: /Users/blah/anaconda2/bin/python
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Found Tclsh: /usr/bin/tclsh (found version "8.5") 
-- Building executables ...
-- OT_EXE_LINKER_FLAGS: OpenTimer;Threads::Threads;stdc++fs
-- TAU15 executable: /Users/blah/vlsi/OpenTimer/bin/ot-tau15
-- Utility executable: /Users/blah/vlsi/OpenTimer/bin/ot-utility
-- OpenTimer shell: /Users/blah/vlsi/OpenTimer/bin/ot-shell
-- Building unit tests ...
-- Building Shell integration tests ...
-- Building TAU15 integration tests ...
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/blah/vlsi/OpenTimer/build

$ make
/Applications/CMake.app/Contents/bin/cmake -H/Users/blah/vlsi/OpenTimer -B/Users/blah/vlsi/OpenTimer/build --check-build-system CMakeFiles/Makefile.cmake 0
/Applications/CMake.app/Contents/bin/cmake -E cmake_progress_start /Users/blah/vlsi/OpenTimer/build/CMakeFiles /Users/blah/vlsi/OpenTimer/build/CMakeFiles/progress.marks
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/Makefile2 all
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/OpenTimer.dir/build.make CMakeFiles/OpenTimer.dir/depend
cd /Users/blah/vlsi/OpenTimer/build && /Applications/CMake.app/Contents/bin/cmake -E cmake_depends "Unix Makefiles" /Users/blah/vlsi/OpenTimer /Users/blah/vlsi/OpenTimer /Users/blah/vlsi/OpenTimer/build /Users/blah/vlsi/OpenTimer/build /Users/blah/vlsi/OpenTimer/build/CMakeFiles/OpenTimer.dir/DependInfo.cmake --color=
Scanning dependencies of target OpenTimer
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/OpenTimer.dir/build.make CMakeFiles/OpenTimer.dir/build
[  2%] Building CXX object CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o
g++   -I/Users/blah/vlsi/OpenTimer  -Wall -O2   -std=c++1z -o CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o -c /Users/blah/vlsi/OpenTimer/ot/unit/unit.cpp
In file included from /Users/blah/vlsi/OpenTimer/ot/unit/unit.cpp:1:
In file included from /Users/blah/vlsi/OpenTimer/ot/unit/unit.hpp:4:
/Users/blah/vlsi/OpenTimer/ot/headerdef.hpp:18:10: fatal error: 'string_view' file not found
#include <string_view>
         ^
1 error generated.
make[2]: *** [CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o] Error 1
make[1]: *** [CMakeFiles/OpenTimer.dir/all] Error 2
make: *** [all] Error 2

twhuang-uiuc, 2018-07-31:
Hi @userghgh , this is due to the compiler version. OpenTimer requires [G++ 7.2](https://gcc.gnu.org/gcc-7/) to build. We do notice this is not correctly captured by our CMakeLists.txt but we just got it fixed. Please update with the master branch and follow the instruction here [Compile OpenTimer](#compile-opentimer).

rk717306, 2018-09-10:
Hello Tsung-Wei, I am facing a similar issue with compiling make. I have installed G++ 7.3.0 version and Clang-5.0 version. It seems to me that something is wrong with clang c++ compiler. I had an issue before with cmake where it could not locate the path of CMAKE_CXX_COMPILER. So i used export to set its path to /usr/bin/clang-5.0 (as in my computer). And then the cmake worked. But now it is giving an error with make. I have included the snapshots of the error. Please let me know if you need more information. Thank you.
[image]
[image]

rk717306, 2018-09-10:
@twhuang-uiuc Thank you for your response. I did  `git pull` but still it is giving me a error. I included the captured images in the comment. Thank you. 
[image]
[image]
[image]

twhuang-uiuc, 2018-09-10:
@rk717306  This seems weird to me. Would you please try one of the following:

1. remove the `build` folder and start over again, or
2. git clone a new OpenTimer from the master branch, or
3. if `clang++` is not working, try `g++` and see if you have problems. You can do this by
```bash
# under OpenTimer project directory
~$ mkdir build
~$ cd build
~$ cmake ../ -DCMAKE_CXX_COMPILER=path/to/your/g++
```

Please let me know.

twhuang-uiuc, 2018-09-10:
@rk717306  I can reproduce your problem under clang++5.0 but everything works ok with clang++6.0. Would you please try upgrading your clang++ to 6.0?

rk717306, 2018-09-11:
Hello @twhuang-uiuc,  I could make it work. Thank you so much for your support all along. I am all excited to start using the tool. Thank you again, I really appreciate your quick responses for my issues.

closed 2018-09-11 by twhuang-uiuc