OpenTimer/OpenTimer #10: [Verilog Parser] Should interpret or ignore attribute syntax (* ... *)

Regor191, 2018-09-05:
Right now the verilog parser seems to be confused by the standard attribute syntax

(* attribute = value *)

It gives '(' cell not found, or something to that effect.

twhuang-uiuc, 2018-09-05:
@Regor191 , right now OpenTimer has a very limited support for verilog syntax - we apologize for this. We are working on a more comprehensive one and hopefully will be included in the next major release. For verilog support, you may visit this [page](https://github.com/OpenTimer/OpenTimer/blob/master/wiki/verilog.md)

Regor191, 2018-09-05:
Thank you for the quick response @twhuang-uiuc , after a little searching my synthesis tool Yosys has an option on the "write_verilog" output command called "-noattr" which will stop it from putting all those attributes out.  So what will solve the problem for me.

closed 2018-09-11 by twhuang-uiuc