OpenTimer/OpenTimer #9: Issue with Installing OpenTimer

rk717306, 2018-09-05:
I did install GNU C++ compiler using Synaptic package manager but it was named as gobjc++ -7 version 7.3.0 16ubunt3 and i also installed clang 5.0,. i could get to the cmake ../ command, but it is giving me a error after that. I have also included a snippet of that error in this message. I googled about it and tried using export to sent the path of GNU c++ compiler. Was unsuccessful doing that as well.

[image]

twhuang-uiuc, 2018-09-05:
Hi @rk717306 , would you please type `g++ -v` from your terminal and let me know what you see? It looks like CMAKE is not able to find your CXX compiler. Would you please also take a look at this [page](https://stackoverflow.com/questions/32801638/cmake-error-at-cmakelists-txt30-project-no-cmake-c-compiler-could-be-found)?

clin99, 2018-09-05:
Hi @rk717306  Could you remove the ```build``` folder and follow below steps to rebuild again?

```
rm -r build
mkdir build
cd build
cmake ../
make
```

Thanks.

twhuang-uiuc, 2018-09-05:
@clin99 Thanks for your reply!
@rk717306 Yes. Please follow that instruction again. Also, it seems you are able to get g++ installed. Would you please type `g++ -v` to see what version you have? You will need g++ at least 7.2 to compile opentimer.