OpenTimer/OpenTimer #11: Internal synchronous paths not reported

Regor191, 2018-09-06:
Internal synchronous (register to register) paths don't seem to get found or reported.

To reproduce, for example, I have a 1-bit scrambler based on linear feedback shift reg,

`module lfs (clk, rst, in1, out1) ;

input clk ;
input rst ;
input in1 ;
output out1 ;

reg out1 ;
reg [11:0]  shift ;

always @(posedge clk or posedge rst) begin
  if (rst) begin
    shift <= 1 ;
	out1 <= 0 ;
  end else begin
    shift <= { shift[10:0], (shift[11]^shift[10])} ;
	out1 <= shift[5] ^ in1 ;
  end
end

endmodule
`
I can synthesize it with the library from the "simple" example and constrain the timing in a similar fashion as that example, and I've inspected the synth output to verify there are many internal paths between bits in the "shift" vector as expected, but those paths don't get reported.  I can report a good number of paths with a fully constrained SDC, but all reported paths involve I/O pins. 

If I then strip out all non-clock constraints so that only the clock signal is constrained, the timing report simply says "no paths found" even though I think the internal register to register delays should be reported with only the clocks defined.

I am using a compile based on a git clone from about Aug. 30 if I recall correctly.

twhuang-uiuc, 2018-09-06:
@Regor191 Would you please share your `.sdc` and `.v` (structural netslit) with me? Since now OpenTimer is an alpha release, it is always a good idea to `git pull` to keep your local up-to-date with the newest master branch. If you are using only `ot-shell`, there should be of no problem in adapting to the new code. I will look into your design and see what I get.

Regor191, 2018-09-06:
[lfs_synth.v.txt](https://github.com/OpenTimer/OpenTimer/files/2357798/lfs_synth.v.txt)
[test.sdc.txt](https://github.com/OpenTimer/OpenTimer/files/2357801/test.sdc.txt)

I had to append the .txt to get the files to attach correctly for some reason.

twhuang-uiuc, 2018-09-10:
Hi @Regor191 , I tried your design with the following commands:

```bash
read_celllib osu018_stdcells.lib
read_verilog lfs.v
read_sdc lfs.sdc
report_path -num_paths 100
```

Based on your given sdc, internal paths are all of positive slacks. This is why you didn't see any negative path reports for register-to-register datapaths. However, if you change your sdc to more strict clock period:

```text
create_clock -name clk1 -period 5 [get_ports clk]
```
You will see the register-to-register paths (with the same commands).

Also, please `git pull` to keep your repo up with the newest master branch. I have improved quite a lot of features and report formats.

Regor191, 2018-09-10:
Thanks.  I misunderstood how the report command was supposed to work.  I'll use a stricter constraint when I'm concerned about having margin.

closed 2018-09-10 by Regor191

twhuang-uiuc, 2018-09-10:
@Regor191 No problem :) Let me know if you have other questions.