OpenTimer/OpenTimer #12: malloc(): memory corruption error using vendor library

Regor191, 2018-09-24:
Using my small lfs.v test design (see internal paths not reported closed issue), I moved implementing with the example library to using a vendor library, the .35um D_CELLS library from XFab, slow 3.0 volt version.  The following bad thing happens : 
  ____              _______              
 / __ \___  ___ ___/_  __(_)_ _  ___ ____
/ /_/ / _ \/ -_) _ \/ / / /  ' \/ -_) __/
\____/ .__/\__/_//_/_/ /_/_/_/_/\__/_/       v2.0.0 (alpha)
    /_/                                     
MIT License: type "license" to see more details.
For help, type "help".
For bug reports, issues, and manual, please see:
<https://github.com/OpenTimer/OpenTimer>.
ot> read_celllib slow.lib
ot> read_verilog lfs_slow.v
ot> read_sdc test.sdc
ot> report_timing
I  9984 18-09-23 22:20:55 verilog.cpp:74] loading netlist "lfs_slow.v" ...
I  1792 18-09-23 22:20:55 celllib.cpp:600] loading celllib "slow.lib" ...
I 14080 18-09-23 22:20:55 sdc.cpp:35] loading sdc "test.sdc" ...
malloc(): memory corruption
Aborted (core dumped)

Unfortunately I believe the library is provided under NDA so I cannot provide it for debug, but it does work with several other tools including Yosys and ABC.  If there is a debug directive or compiler flag that I can turn on to narrow down where this crash happened, let me know how it is used and I'll try to do it.
I did do the "git pull" right before this run as you had suggested.  Also, the the only difference between this and what was working before is the cell library used for synthesis and timing.  
I am running Ubuntu 18.04.1 LTS.

twhuang-uiuc, 2018-09-24:
Hi @Regor191 , is there any way you can generate a representative library for me so I can test it on my side? I know there is NDA but if you are able to create a "fake" library with similar syntax & structure, it will be easier for me to help you out. 

Or can can run your program with gdb or valgrind and show me the error message?

Regor191, 2018-09-24:
gdb sounded like the easiest option, so I tried that but it says there was no debug info, and the crash didn't show much more information.  I did a stack trace and it did seem to know where the calls are coming from.  Is there an easy way to tell CMake to add debug info?  Here are the top 12 items on the stack trace : 

#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff6ec1801 in __GI_abort () at abort.c:79
#2  0x00007ffff6f0a897 in __libc_message (action=action@entry=do_abort, 
    fmt=fmt@entry=0x7ffff7037b9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff6f1190a in malloc_printerr (
    str=str@entry=0x7ffff7035e0e "malloc(): memory corruption")
    at malloc.c:5350
#4  0x00007ffff6f15994 in _int_malloc (av=av@entry=0x7fffe0000020, 
    bytes=bytes@entry=31) at malloc.c:3738
#5  0x00007ffff6f182ed in __GI___libc_malloc (bytes=31) at malloc.c:3065
#6  0x00007ffff78bb54c in operator new(unsigned long) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x0000555555597d47 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_mutate(unsigned long, unsigned long, char const*, unsigned long) ()
#8  0x000055555559d34b in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace(unsigned long, unsigned long, char const*, unsigned long) ()
#9  0x00005555555c1f41 in ot::Celllib::_extract_lut(__gnu_cxx::__normal_iterator<std::basic_string_view<char, std::char_traits<char> >*, std::vector<std::basic_string_view<char, std::char_traits<char> >, std::allocator<std::basic_string_view<char, std::char_traits<char> > > > >&, __gnu_cxx::__normal_iterator<std::basic_string_view<char, std::char_traits<char> >*, std::vector<std::basic_string_view---Type <return> to continue, or q <return> to quit---
<char, std::char_traits<char> >, std::allocator<std::basic_string_view<char, std::char_traits<char> > > > >) ()
#10 0x00005555555c36e0 in ot::Celllib::_extract_timing(__gnu_cxx::__normal_iterator<std::basic_string_view<char, std::char_traits<char> >*, std::vector<std::basic_string_view<char, std::char_traits<char> >, std::allocator<std::basic_string_view<char, std::char_traits<char> > > > >&, __gnu_cxx::__normal_iterator<std::basic_string_view<char, std::char_traits<char> >*, std::vector<std::basic_string_view<char, std::char_traits<char> >, std::allocator<std::basic_string_view<char, std::char_traits<char> > > > >) ()
#11 0x00005555555c46a8 in ot::Celllib::_extract_cellpin(__gnu_cxx::__normal_iterator<std::basic_string_view<char, std::char_traits<char> >*, std::vector<std::basic_string_view<char, std::char_traits<char> >, std::allocator<std::basic_string_view<char, std::char_traits<char> > > > >&, __gnu_cxx::__normal_iterator<std::basic_string_view<char, std::char_traits<char> >*, std::vector<std::basic_string_view<char, std::char_traits<char> >, std::allocator<std::basic_string_view<char, std::char_traits<char> > > > >) ()
#12 0x00005555555c54b0 in ot::Celllib::_extract_cell(__gnu_cxx::__normal_iterator<std::basic_string_view<char, std::char_traits<char> >*, std::vector<std::basic_string_view<char, std::char_traits<char> >, std::allocator<std::basic_string_view<char, std::char_traits<char> > > > >&, __gnu_cxx::__normal_iterator<std::basic_string_view<char, std::char_traits<char> >*, std::vector<std::basic_string_view<char, std::char_traits<char> >, std::allocator<std::basic_string_view<char, std::char_traits<char> > > > >) ()

Regor191, 2018-09-24:
Here's another stack trace with the debug info compiled in.
It looks like celllib.cpp line 264 might be where it crashed?  There are some variable states in there that you might be able to use as well.
[crash1.txt](https://github.com/OpenTimer/OpenTimer/files/2412371/crash1.txt)

twhuang-uiuc, 2018-09-25:
Hi @Regor191  I may be a bit delay on this because I am now on travel. But I will get into this as soon as possible. Please apologize my late response if any.

Regor191, 2018-09-26:
I've been trying to find the problem but I'm more of an old-school c guy so c++ is a little hard for me.  But I have gathered more information that you might find useful.

1. The liberty file in the failing case appears to have good syntax all around the area that is being parsed when the crash occurs.  It has 7x7 lookup tables as opposed to the 5x5 tables in the examples.  It also uses templates.  
2. Memory corruption happens after about 11 executions of _extract_lut, independent of where this happens to fall in the library file.  I can edit the library file to remove the exact timing tables where it previously crashed, and it will crash on a completely different table after about the same number of _extract_lut executions. 
3.  Additional logging shows that the tokens and arguments around the crash are normal, and the memory corruption is generally detected deep in a standard c library call.  Sometimes the logging function itself will detect the memory corruption and crash out when this is added.
4.  To check for lack of available memory, I added (at the top of _extract_lut) a malloc / free with NULL check on the malloc.  The result was the malloc succeeds but the free command crashes on the second call to _extract_lut.  It doesn't matter whether I malloc 10,000 byes or 3000 bytes, it crashes in the same place.  
5.  The size of the liberty file doesn't seem to matter much.  I cut it from 16M down to 1.5M by removing most of the parts, but it fails in the same way.
6.  Updating g++ from 7.3.0 to 8.0.1 made no difference.

Enjoy your travels and I hope we can figure this out when you get back.

Regor191, 2018-09-26:
I think I found the problem.  The _extract_lut code doesn't handle sizing the lut.table correctly when the indices are only in the template, and not included in the local timing instance.  In that case, the table will be resized to 1, but the full value table will be read with pointer increment i++, which corrupts the memory.
I don't know if the other parts of the code are relying on the indices to be copied into the lut structure from the template or not, but if so this might be a good place to do it, if it doesn't happen elsewhere.

3ll34ndr0, 2018-09-26:
Hi Regor,
once I had a similar situation. I found that if I also load a spef file for
that netlist, the report_timing command didn't crashed. If you happen to
have a spef for that circuit, maybe you can try and see what happen.
Regards,
Leandro.

twhuang-uiuc, 2018-09-26:
Hi @Regor191 , the indices are required for inter- and extrapolation. For example, the below example is a complete timing LUT. OpenTimer will read `index_1` and `index_2` to figure out the table size and read in the values iteratively. Could you please confirm if in your case this is the problem?

```text
fall_transition ("delay_outputslew_template_7X8") {
         index_1 ("0.00,1.00,2.00,4.00,8.00,16.00,32.00") ;
         index_2 ("5.00,30.00,50.00,80.00,140.00,200.00,300.00,500.00") ;
         values (\
         "10, 10.976, 13.104, 16.08, 20.136, 22.92, 26.36, 31.864",\
         "15, 15.36, 16.92, 20.224, 25.72, 29.648, 34.384, 41.048",\
         "20, 20.072, 21.128, 23.928, 30.376, 35.272, 41.328, 49.488",\
         "30, 30, 30.256, 32.08, 38.128, 44.616, 52.912, 64.456",\
         "50, 50, 50, 50.32, 54.008, 59.808, 71.024, 88.184",\
         "90, 90, 90, 90, 90.448, 93.336, 101.536, 123.584",\
         "170, 170, 170, 170, 170, 170, 172.12, 185.672"\
         );
        }
```

Regor191, 2018-09-26:
The indices are not included locally in every cell, they are only in the template near the beginning of the file.

It is this syntax : 

  lu_table_template (TIMING_TEMP_0) {
    variable_1 : input_net_transition;
    variable_2 : total_output_net_capacitance;
    index_1 ("0.00,1.00,2.00,4.00,8.00,16.00,32.00") ;
    index_2 ("5.00,30.00,50.00,80.00,140.00,200.00,300.00,500.00") ;
  }

... a bunch of other stuff ...

fall_transition (TIMING_TEMP_0) {
         values (\
         "10, 10.976, 13.104, 16.08, 20.136, 22.92, 26.36, 31.864",\
         "15, 15.36, 16.92, 20.224, 25.72, 29.648, 34.384, 41.048",\
         "20, 20.072, 21.128, 23.928, 30.376, 35.272, 41.328, 49.488",\
         "30, 30, 30.256, 32.08, 38.128, 44.616, 52.912, 64.456",\
         "50, 50, 50, 50.32, 54.008, 59.808, 71.024, 88.184",\
         "90, 90, 90, 90, 90.448, 93.336, 101.536, 123.584",\
         "170, 170, 170, 170, 170, 170, 172.12, 185.672"\
         );
        }

twhuang-uiuc, 2018-09-26:
Hi @Regor191  I see. I will add a block to initialize the table size from the lut template if no indices are found in the local table definition block. Will let u know when this is done soon.

Regor191, 2018-09-26:
I was able to get my copy working by pre-loading the template indices before the main while loop and updating the sizes when the template pointer is non-null.  However I probably broke it for the other syntax for now... I would need to clear out the pre-loaded indices when the local indices are there otherwise they will append instead of overwriting.  
Another good feature to add around this would be to check that i == size1*size2 after the table read and fail out if it is not.
If there is some syntax for loading the table that won't corrupt the memory if the table is too big, that might be better also, but I'm too much of a c++ novice to suggest anything specific.

twhuang-uiuc, 2018-09-26:
Hi @Regor191 Not a problem at all. In fact, you are giving very good suggestion and bug report. I will fix it as soon as I finish the conference. At the same time, please let me know any other issues you have. I will let you know when I work it out.