OpenTimer/OpenTimer #14: tcl based opentimer runs

kunalg123, 2018-09-26:
Hi Tsung-Wei
I noticed another issue while running with tcl based Opentimer
To reproduce the issue, use the below steps
1) tar -xvzf testcase.tar.gz
2) tclsh test.tcl
3) ot-shell -i test.conf

Output of step 2 is redirected in my_report file and output of step 3 is on shell
Both outputs should be identical, but they are not.
Step 2 drops some constraints while step 3 is accurate. Can you please have a look?

kunalg123, 2018-09-27:
Hi
I hope you have received the testcase?

kunalg123, 2018-09-27:
Uploaded
Can you please check if you are able to view it?

twhuang-uiuc, 2018-10-03:
Hi @kunalg123 , I am working on this - just came back from my trip. Sorry for the waiting!

clin99, 2018-10-03:
Hi @kunalg123 , I follow your steps and the outputs of step 2 and 3 are below:

**Step 2 (my_report):**
```
I 14080 18-10-03 15:41:38 sdc.cpp:35] loading sdc "map9v3.opentimer.sdc" ...                                                           
I 26368 18-10-03 15:41:38 celllib.cpp:600] loading celllib "osu018/osu018_stdcells.lib" ... 
I 22272 18-10-03 15:41:38 celllib.cpp:600] loading celllib "osu018/osu018_stdcells.lib" ...
I 18176 18-10-03 15:41:38 verilog.cpp:74] loading netlist "map9v3.final.synth.v" ...
I 22272 18-10-03 15:41:38 unit.cpp:244] use celllib time unit 1e-09 s
I 22272 18-10-03 15:41:38 unit.cpp:258] use celllib capacitance unit 1e-12 F
I 22272 18-10-03 15:41:38 unit.cpp:272] use celllib current unit 1e-06 A
I 22272 18-10-03 15:41:38 unit.cpp:286] use celllib voltage unit 1 V
I 22272 18-10-03 15:41:38 unit.cpp:300] use celllib resistance unit 1000 Ohm
I 22272 18-10-03 15:41:38 unit.cpp:314] use celllib power unit 1e-09 W 
I 22272 18-10-03 15:41:38 celllib.cpp:69] added min celllib "osu018_stdcells" [cells:32]
I 22272 18-10-03 15:41:38 celllib.cpp:69] added max celllib "osu018_stdcells" [cells:32]
I 22272 18-10-03 15:41:38 verilog.cpp:21] added verilog module "map9v3" [gates:209]
/home/clin99/Software/temp/OpenTimer/bin/map9v3.opentimer.sdc, 191: incorrect set of required parameters for "set_output_delay"
I 14080 18-10-03 15:41:39 sdc.cpp:21] added 188 sdc commands
Startpoint    : N_6_
Endpoint      : DFFSR_21:D
Analysis type : min 
------------------------------------------------------
       Type       Delay        Time   Dir  Description
------------------------------------------------------
       port     102.000     102.000  rise  N_6_
        pin       0.000     102.000  rise  NAND2X1_8:A (NAND2X1)
        pin      -8.865      93.135  fall  NAND2X1_8:Y (NAND2X1)
        pin       0.000      93.135  fall  NAND2X1_9:B (NAND2X1)
        pin       0.119      93.254  rise  NAND2X1_9:Y (NAND2X1)
        pin       0.000      93.254  rise  OAI21X1_21:B (OAI21X1)
        pin       0.049      93.303  fall  OAI21X1_21:Y (OAI21X1)
        pin       0.000      93.303  fall  DFFSR_21:D (DFFSR)
    arrival                  93.303        data arrival time

related pin     153.068     153.068  rise  DFFSR_21:CLK (DFFSR)
 constraint       0.130     153.198        library hold_rising
   required                 153.198        data required time
------------------------------------------------------
      slack                 -59.895        VIOLATED
-59.9
```

**Step 3:**
```
I 55040 18-10-03 15:42:48 verilog.cpp:74] loading netlist "map9v3.final.synth.v" ...
I  1792 18-10-03 15:42:48 sdc.cpp:35] loading sdc "map9v3.opentimer.sdc" ...
I 63232 18-10-03 15:42:48 celllib.cpp:600] loading celllib "osu018/osu018_stdcells.lib" ...
I 59136 18-10-03 15:42:48 celllib.cpp:600] loading celllib "osu018/osu018_stdcells.lib" ...
I 63232 18-10-03 15:42:48 unit.cpp:244] use celllib time unit 1e-09 s
I 63232 18-10-03 15:42:48 unit.cpp:258] use celllib capacitance unit 1e-12 F
I 63232 18-10-03 15:42:48 unit.cpp:272] use celllib current unit 1e-06 A
I 63232 18-10-03 15:42:48 unit.cpp:286] use celllib voltage unit 1 V
I 63232 18-10-03 15:42:48 unit.cpp:300] use celllib resistance unit 1000 Ohm
I 63232 18-10-03 15:42:48 unit.cpp:314] use celllib power unit 1e-09 W
I 63232 18-10-03 15:42:48 celllib.cpp:69] added min celllib "osu018_stdcells" [cells:32]
I 63232 18-10-03 15:42:48 celllib.cpp:69] added max celllib "osu018_stdcells" [cells:32]
I 63232 18-10-03 15:42:48 verilog.cpp:21] added verilog module "map9v3" [gates:209]
I  1792 18-10-03 15:42:48 sdc.cpp:21] added 227 sdc commands
Startpoint    : N_6_
Endpoint      : DFFSR_21:D
Analysis type : min
------------------------------------------------------
       Type       Delay        Time   Dir  Description
------------------------------------------------------
       port     102.000     102.000  rise  N_6_
        pin       0.000     102.000  rise  NAND2X1_8:A (NAND2X1)
        pin      -8.865      93.135  fall  NAND2X1_8:Y (NAND2X1)
        pin       0.000      93.135  fall  NAND2X1_9:B (NAND2X1)
        pin       0.119      93.254  rise  NAND2X1_9:Y (NAND2X1)
        pin       0.000      93.254  rise  OAI21X1_21:B (OAI21X1)
        pin       0.049      93.303  fall  OAI21X1_21:Y (OAI21X1)
        pin       0.000      93.303  fall  DFFSR_21:D (DFFSR)
    arrival                  93.303        data arrival time

related pin     153.068     153.068  rise  DFFSR_21:CLK (DFFSR)
 constraint       0.130     153.198        library hold_rising
   required                 153.198        data required time
------------------------------------------------------
      slack                 -59.895        VIOLATED
-59.9
```

The difference is that **Step2** shows a message during parsing the sdc file (the parser is provided by Synopsys):
```
/home/clin99/Software/temp/OpenTimer/bin/map9v3.opentimer.sdc, 191: incorrect set of required parameters for "set_output_delay"
```

The ``map9v3.opentimer.sdc`` file is created by ``procs/read_sdc.proc`` script. Could you please elaborate how the script generates the sdc file?

Thanks!

twhuang-uiuc, 2018-10-04:
@kunalg123  I suspect there is inconsistency between the two sdc files you fed to OpenTimer. If you look at the log, one takes 227 and the other takes 188 sdc commands.

kunalg123, 2018-10-04:
Hi 
The sdc file is created from TCL proc
The sdc files (if you look into my_report file, and open sdc file pointed there) in both cases are same. While generating and running ot-shell using TCL script, for some reason it drops constraints post line number 191 from sdc file. 
Do you have some debug commands in ot-shell to find out, why its dropping constraints ?

twhuang-uiuc, 2018-10-04:
Hi @kunalg123 , have you tried running it with `gdb` or `valgrind`? We use Synopsys open-source Tap-in SDC parser and the error message you saw was generated by it internally. Can you try dumping out the sdc contents before feeding it to opentimer (for both cases)? Ideally they should be the same.

twhuang-uiuc, 2018-10-05:
Can you paste the log message you saw from your terminal?

kunalg123, 2018-10-05:
kunalg@kunalg-VirtualBox ~/Desktop/work/tools/vsdflow/vsdflow-master/vsdflow_standalone_testing/map9v3/testcase $ tclsh test.tcl 

Info: Timing Analysis Started....

Info: Initializing number of threads, libraries, sdc, verilog netlist path...
input conf file is test.conf
output report file is my_report
Startpoint    : N_6_
Endpoint      : DFFSR_21:D
Analysis type : min
------------------------------------------------------
       Type       Delay        Time   Dir  Description
------------------------------------------------------
       port     102.000     102.000  rise  N_6_
        pin       0.000     102.000  rise  NAND2X1_8:A (NAND2X1)
        pin      -8.865      93.135  fall  NAND2X1_8:Y (NAND2X1)
        pin       0.000      93.135  fall  NAND2X1_9:B (NAND2X1)
        pin       0.119      93.254  rise  NAND2X1_9:Y (NAND2X1)
        pin       0.000      93.254  rise  OAI21X1_21:B (OAI21X1)
        pin       0.049      93.303  fall  OAI21X1_21:Y (OAI21X1)
        pin       0.000      93.303  fall  DFFSR_21:D (DFFSR)
    arrival                  93.303        data arrival time

related pin     153.068     153.068  rise  DFFSR_21:CLK (DFFSR)
 constraint       0.130     153.198        library hold_rising
   required                 153.198        data required time
------------------------------------------------------
      slack                 -59.895        VIOLATED
-59.9
I 22272 18-10-05 20:22:12 sdc.cpp:35] loading sdc "map9v3.opentimer.sdc" ...
/home/kunalg/Desktop/work/tools/vsdflow/vsdflow-master/vsdflow_standalone_testing/map9v3/testcase/map9v3.opentimer.sdc, 191: incorrect set of required parameters for "set_output_delay"
I 22272 18-10-05 20:22:13 verilog.cpp:74] loading netlist "map9v3.final.synth.v" ...
I 22272 18-10-05 20:22:13 celllib.cpp:600] loading celllib "/usr/local/share/qflow/tech/osu018/osu018_stdcells.lib" ...
I 22272 18-10-05 20:22:13 celllib.cpp:600] loading celllib "/usr/local/share/qflow/tech/osu018/osu018_stdcells.lib" ...
I 22272 18-10-05 20:22:13 unit.cpp:244] use celllib time unit 1e-09 s
I 22272 18-10-05 20:22:13 unit.cpp:258] use celllib capacitance unit 1e-12 F
I 22272 18-10-05 20:22:13 unit.cpp:272] use celllib current unit 1e-06 A
I 22272 18-10-05 20:22:13 unit.cpp:286] use celllib voltage unit 1 V
I 22272 18-10-05 20:22:13 unit.cpp:300] use celllib resistance unit 1000 Ohm
I 22272 18-10-05 20:22:13 unit.cpp:314] use celllib power unit 1e-09 W
I 22272 18-10-05 20:22:13 celllib.cpp:69] added min celllib "osu018_stdcells" [cells:32]
I 22272 18-10-05 20:22:13 celllib.cpp:69] added max celllib "osu018_stdcells" [cells:32]
I 22272 18-10-05 20:22:13 verilog.cpp:21] added verilog module "map9v3" [gates:209]
I 22272 18-10-05 20:22:13 sdc.cpp:21] added 188 sdc commands
    while executing
"exec /home/kunalg/Desktop/work/tools/opentimer/OpenTimer/bin/ot-shell -i $inputConfFile"
    (file "test.tcl" line 51)
kunalg@kunalg-VirtualBox ~/Desktop/work/tools/vsdflow/vsdflow-master/vsdflow_standalone_testing/map9v3/testcase $

twhuang-uiuc, 2018-10-05:
Hi @kunalg123  I suspected the error comes from the way you generate these sdc during tcl process. Maybe the file didn't flush out completely before you call OpenTimer to read sdc. This may be the reason it only gets 188 sdc commands?

twhuang-uiuc, 2018-10-06:
@kunalg123 , no apology at all! Please let us know if you run into more issues.