OpenTimer/OpenTimer #16: opentimer aborts with cppr -enable

kunalg123, 2018-10-04:
Hi 
I added "cppr -enable" command in map9v3 testcase, which comes from opentimer installation (along with full fledged map9v3 sdc constraints), it aborts giving the below message 
OpenTimer: /home/kunalg/Desktop/work/tools/opentimer/OpenTimer/ot/timer/path.cpp:192: ot::Path::dump(std::ostream&) const::<lambda(ot::Test*)>: Assertion `std::fabs(sum - rat) < 1e-2f' failed.
Aborted
I have attached the testcase. To reproduce, use the below steps
1) tar -xvzf map9v3.tar.gz
2) cd map9v3
3) ot-shell -i map9v3.conf

twhuang-uiuc, 2018-10-04:
Hi @kunalg123  I am not able to see the file you uploaded. Would you please double check?

kunalg123, 2018-10-05:
Are you able to see it now?

kunalg123, 2018-10-05:
Resending the testcase
Not sure, whether its getting uploaded or not on github

kunalg123, 2018-10-05:
file:///home/kunalg/Desktop/work/tools/vsdflow/vsdflow-master/vsdflow_standalone_testing/map9v3/testcase.tar.gz

kunalg123, 2018-10-06:
Hi Tsung-Wei
For some reasons, I am not able to reproduce this issue with any of my other testcases. So closing this one
Will open it if I face similar issues

closed 2018-10-06 by kunalg123