OpenTimer/OpenTimer #19: Mac OS clang string is AppleClang so I change cmake compiler check from "Clang" to "AppleClang"

xianaiyang, 2018-11-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

xianaiyang, 2018-11-01:
but has <experiment/system> file not found issue

xianaiyang, 2018-11-14:
thanks for your reply. I can compile OpenTimer after I change compiler to g++7.3 
but with latest code I will got following compile issue
taskflow.hpp:232:7: error: use of deleted function 'std::atomic<int>::atomic(const std::atomic<int>&)'

twhuang-uiuc, 2018-11-14:
Hi @xianaiyang , turns out it is a GCC bug as described [here](https://stackoverflow.com/questions/48991967/container-of-non-copyable-in-stdoptional?fbclid=IwAR34fEd0wdYrIFEgs5c03rh1PimCkcPEXcjsaWfYYHHNkvtWZvwZmYSbIIo). Would you please try updating to gcc 8? 

Although I didn't see any problem here in my Linux machine, I do reproduce this problem at my MAC OSX. Are you using Linux machines?

I will add a compiler statement later in the `CMakeLists.txt` to enforce this constraint.

xianaiyang, 2018-11-15:
I am using mac osx

twhuang-uiuc, 2018-11-17:
Hi @xianaiyang  Would you try g++8? On mac you will need `sudo port install gcc8`. I tried it and it worked. Also make sure you select the right version during cmake

```bash
~$ cmake ../ -DCMAKE_CXX_COMPILER=/path/to/your/g++8
~$ make
```

xianaiyang, 2018-11-19:
Thanks I tryed gcc8.0 it works well.

twhuang-uiuc, 2018-11-19:
@xianaiyang Thank you! And sorry for the inconvenience!

hughperkins, 2022-03-07:
Perhaps we could updaet the doc to point out that we should use gcc, instead of clang?