OpenTimer/OpenTimer #21: [Compilation Issue] Problem with compiling OpenTimer on Mac

geochrist, 2018-11-26:
I’m getting the following error during compilation on my Mac.  Any ideas what might be wrong? 

Scanning dependencies of target OpenTimer
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/OpenTimer.dir/build.make CMakeFiles/OpenTimer.dir/build
[  1%] Building CXX object CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o
/opt/local/bin/g++-mp-7   -I/Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master  -Wall -O2   -std=c++1z -o CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o -c /Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/unit/unit.cpp
In file included from /opt/local/include/gcc7/c++/x86_64-apple-darwin17/bits/c++allocator.h:33:0,
                 from /opt/local/include/gcc7/c++/bits/allocator.h:46,
                 from /opt/local/include/gcc7/c++/string:41,
                 from /opt/local/include/gcc7/c++/bits/locale_classes.h:40,
                 from /opt/local/include/gcc7/c++/bits/ios_base.h:41,
                 from /opt/local/include/gcc7/c++/ios:42,
                 from /opt/local/include/gcc7/c++/ostream:38,
                 from /opt/local/include/gcc7/c++/iostream:39,
                 from /Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/headerdef.hpp:4,
                 from /Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/unit/unit.hpp:4,
                 from /Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/unit/unit.cpp:1:
/opt/local/include/gcc7/c++/ext/new_allocator.h: In instantiation of 'void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = tf::Node; _Args = {const tf::Node&}; _Tp = tf::Node]':
/opt/local/include/gcc7/c++/bits/alloc_traits.h:475:4:   required from 'static void std::allocator_traits<std::allocator<_CharT> >::construct(std::allocator_traits<std::allocator<_CharT> >::allocator_type&, _Up*, _Args&& ...) [with _Up = tf::Node; _Args = {const tf::Node&}; _Tp = tf::Node; std::allocator_traits<std::allocator<_CharT> >::allocator_type = std::allocator<tf::Node>]'
/opt/local/include/gcc7/c++/bits/forward_list.h:351:32:   required from 'std::_Fwd_list_base<_Tp, _Alloc>::_Node* std::_Fwd_list_base<_Tp, _Alloc>::_M_create_node(_Args&& ...) [with _Args = {const tf::Node&}; _Tp = tf::Node; _Alloc = std::allocator<tf::Node>; std::_Fwd_list_base<_Tp, _Alloc>::_Node = std::_Fwd_list_node<tf::Node>]'
/opt/local/include/gcc7/c++/bits/forward_list.tcc:109:27:   required from 'void std::forward_list<_Tp, _Alloc>::_M_range_initialize(_InputIterator, _InputIterator) [with _InputIterator = std::_Fwd_list_const_iterator<tf::Node>; _Tp = tf::Node; _Alloc = std::allocator<tf::Node>]'
/opt/local/include/gcc7/c++/bits/forward_list.h:531:28:   required from 'std::forward_list<_Tp, _Alloc>::forward_list(const std::forward_list<_Tp, _Alloc>&) [with _Tp = tf::Node; _Alloc = std::allocator<tf::Node>]'
/opt/local/include/gcc7/c++/type_traits:1406:12:   required from 'struct std::is_trivially_copy_constructible<std::forward_list<tf::Node> >'
/opt/local/include/gcc7/c++/optional:105:8:   required from 'class std::_Optional_base<std::forward_list<tf::Node> >'
/opt/local/include/gcc7/c++/optional:453:11:   required from 'class std::optional<std::forward_list<tf::Node> >'
/Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/taskflow/taskflow.hpp:267:26:   required from here
/opt/local/include/gcc7/c++/ext/new_allocator.h:136:4: error: use of deleted function 'tf::Node::Node(const tf::Node&)'
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/headerdef.hpp:58:0,
                 from /Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/unit/unit.hpp:4,
                 from /Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/unit/unit.cpp:1:
/Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/taskflow/taskflow.hpp:232:7: note: 'tf::Node::Node(const tf::Node&)' is implicitly deleted because the default definition would be ill-formed:
 class Node {
       ^~~~
/Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/taskflow/taskflow.hpp:232:7: error: use of deleted function 'std::atomic<int>::atomic(const std::atomic<int>&)'
In file included from /opt/local/include/gcc7/c++/future:42:0,
                 from /Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/headerdef.hpp:21,
                 from /Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/unit/unit.hpp:4,
                 from /Users/gjchen/src/OpenTimer-v2.181125/OpenTimer-master/ot/unit/unit.cpp:1:
/opt/local/include/gcc7/c++/atomic:668:7: note: declared here
       atomic(const atomic&) = delete;
       ^~~~~~
make[2]: *** [CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o] Error 1
make[1]: *** [CMakeFiles/OpenTimer.dir/all] Error 2
make: *** [all] Error 2

twhuang-uiuc, 2018-11-26:
Hi @geochrist 

This is a g++ bug found in 7.2 series for the C++17 `std::optional`. In some platforms, v7.3 got it fixed but mostly likely you will need to upgrade your g++ to v8.0. Would you please try:

```bash
~$ sudo port install gcc8  
```
or
```bash
~$ brew install gcc@8
```
to make OpenTimer v2 compiled under g++ v8? I will later on add a requirement on the compiler version in the CMakeList. Sorry for this inconvenience.

geochrist, 2018-11-26:
Works great!

closed 2018-11-26 by geochrist