OpenTimer/OpenTimer #24: [Verilog Issue] declare multiple nets in a single statement

kat54347, 2018-12-03:
**Describe the bug**
I get the error when I declare more than one variables in a single statement in my Verilog file
 `E 59136 18-12-03 15:43:27 timer.cpp:160] cell n2 not found in celllib`

**To Reproduce**
Using the provided example simple.v for convenience.

Steps to reproduce the behavior:
1. Modify simple.v to declare two wires in a single statement:
```
module simple (
inp1,
inp2,
tau2015_clk,
out
);

// Start PIs
input inp1;
input inp2;
input tau2015_clk;

// Start POs
output out;

// Start wires
wire n1, n2;
wire n3;
wire n4;
wire inp1;
wire inp2;
wire tau2015_clk;
wire out;

// Start cells
NAND2X1 u1 ( .A(inp1), .B(inp2), .Y(n1) );
DFFNEGX1 f1 ( .D(n2), .CLK(tau2015_clk), .Q(n3) );
INVX1 u2 ( .A(n3), .Y(n4) );
INVX2 u3 ( .A(n4), .Y(out) );
NOR2X1 u4 ( .A(n1), .B(n3), .Y(n2) );

endmodule
```

2. Run ot-shell with provided simple.conf

3. See error
```
E 26368 18-12-03 15:48:17 timer.cpp:160] cell n2 not found in celllib
I 26368 18-12-03 15:48:17 verilog.cpp:24] added verilog module "simple" [gates:5]
```

**Expected behavior**
Expected OpenTimer to parse net/variable declarations correctly

**Desktop (please complete the following information):**
 - OS: Springdale Linux  release 6.10

twhuang-uiuc, 2018-12-03:
Hi @kat54347 , this is the verilog parser issue. Would you please change the line `wire n1, n2;` to two separate lines:

```
wire n1;
wire n2;
```

Sorry for this inconvenience, we are still working on the verilog parser to enhance this capability.

ShantanuRaoo, 2024-11-16:
i am parsing each wire in different line still getting error  each time
17:44:27 verilog.cpp:160] missing ; in instance declaration
however there is no such ; missing