OpenTimer/OpenTimer #29: Compile error with clang++

renau, 2019-04-20:
I get this compile errors/warnings with the latest clang++ 
clang version 8.0.0 (tags/RELEASE_800/final)

external/opentimer/ot/timer/timer.cpp:1109:77: error: 'el' in capture list does not name a variable
    _insert_action(to_string("update_endpoints ", el, ' ', rf)).work([this, el, rf] () {

external/opentimer/ot/timer/timer.cpp:1109:81: error: 'rf' in capture list does not name a variable
    _insert_action(to_string("update_endpoints ", el, ' ', rf)).work([this, el, rf] () {
                                                                                ^
external/opentimer/ot/timer/timer.cpp:1113:28: error: reference to local binding 'el' declared in enclosing function 'ot::Timer::_update_endpoints'
        if(po.second.slack(el, rf).has_value()) {
                           ^
external/opentimer/ot/timer/timer.cpp:1105:18: note: 'el' declared here
  FOR_EACH_EL_RF(el, rf) {
                 ^
external/opentimer/ot/timer/timer.cpp:1113:32: error: reference to local binding 'rf' declared in enclosing function 'ot::Timer::_update_endpoints'
        if(po.second.slack(el, rf).has_value()) {
                               ^
external/opentimer/ot/timer/timer.cpp:1105:22: note: 'rf' declared here
  FOR_EACH_EL_RF(el, rf) {
                     ^
external/opentimer/ot/timer/timer.cpp:1114:22: error: reference to local binding 'el' declared in enclosing function 'ot::Timer::_update_endpoints'
          _endpoints[el][rf].emplace_back(el, rf, po.second);
                     ^
external/opentimer/ot/timer/timer.cpp:1105:18: note: 'el' declared here
  FOR_EACH_EL_RF(el, rf) {
                 ^
external/opentimer/ot/timer/timer.cpp:1114:26: error: reference to local binding 'rf' declared in enclosing function 'ot::Timer::_update_endpoints'
          _endpoints[el][rf].emplace_back(el, rf, po.second);
                         ^
external/opentimer/ot/timer/timer.cpp:1105:22: note: 'rf' declared here
  FOR_EACH_EL_RF(el, rf) {
                     ^
external/opentimer/ot/timer/timer.cpp:1114:43: error: reference to local binding 'el' declared in enclosing function 'ot::Timer::_update_endpoints'
          _endpoints[el][rf].emplace_back(el, rf, po.second);
                                          ^
external/opentimer/ot/timer/timer.cpp:1105:18: note: 'el' declared here
  FOR_EACH_EL_RF(el, rf) {
                 ^
external/opentimer/ot/timer/timer.cpp:1114:47: error: reference to local binding 'rf' declared in enclosing function 'ot::Timer::_update_endpoints'
          _endpoints[el][rf].emplace_back(el, rf, po.second);
                                              ^
external/opentimer/ot/timer/timer.cpp:1105:22: note: 'rf' declared here
  FOR_EACH_EL_RF(el, rf) {
                     ^
external/opentimer/ot/timer/timer.cpp:1120:23: error: reference to local binding 'el' declared in enclosing function 'ot::Timer::_update_endpoints'
        if(test.slack(el, rf).has_value()) {
                      ^
external/opentimer/ot/timer/timer.cpp:1105:18: note: 'el' declared here
  FOR_EACH_EL_RF(el, rf) {
                 ^
external/opentimer/ot/timer/timer.cpp:1120:27: error: reference to local binding 'rf' declared in enclosing function 'ot::Timer::_update_endpoints'
        if(test.slack(el, rf).has_value()) {
                          ^
external/opentimer/ot/timer/timer.cpp:1105:22: note: 'rf' declared here
  FOR_EACH_EL_RF(el, rf) {
                     ^
external/opentimer/ot/timer/timer.cpp:1121:22: error: reference to local binding 'el' declared in enclosing function 'ot::Timer::_update_endpoints'
          _endpoints[el][rf].emplace_back(el, rf, test);
                     ^
external/opentimer/ot/timer/timer.cpp:1105:18: note: 'el' declared here
  FOR_EACH_EL_RF(el, rf) {
                 ^
external/opentimer/ot/timer/timer.cpp:1121:26: error: reference to local binding 'rf' declared in enclosing function 'ot::Timer::_update_endpoints'
          _endpoints[el][rf].emplace_back(el, rf, test);
                         ^
external/opentimer/ot/timer/timer.cpp:1105:22: note: 'rf' declared here
  FOR_EACH_EL_RF(el, rf) {
                     ^
external/opentimer/ot/timer/timer.cpp:1121:43: error: reference to local binding 'el' declared in enclosing function 'ot::Timer::_update_endpoints'
          _endpoints[el][rf].emplace_back(el, rf, test);
                                          ^
external/opentimer/ot/timer/timer.cpp:1105:18: note: 'el' declared here
  FOR_EACH_EL_RF(el, rf) {
                 ^
external/opentimer/ot/timer/timer.cpp:1121:47: error: reference to local binding 'rf' declared in enclosing function 'ot::Timer::_update_endpoints'
          _endpoints[el][rf].emplace_back(el, rf, test);
                                              ^
external/opentimer/ot/timer/timer.cpp:1105:22: note: 'rf' declared here
  FOR_EACH_EL_RF(el, rf) {
                     ^
external/opentimer/ot/timer/timer.cpp:1126:28: error: reference to local binding 'el' declared in enclosing function 'ot::Timer::_update_endpoints'
      std::sort(_endpoints[el][rf].begin(), _endpoints[el][rf].end());
                           ^
external/opentimer/ot/timer/timer.cpp:1105:18: note: 'el' declared here
  FOR_EACH_EL_RF(el, rf) {
                 ^
external/opentimer/ot/timer/timer.cpp:1126:32: error: reference to local binding 'rf' declared in enclosing function 'ot::Timer::_update_endpoints'
      std::sort(_endpoints[el][rf].begin(), _endpoints[el][rf].end());
                               ^
external/opentimer/ot/timer/timer.cpp:1105:22: note: 'rf' declared here
  FOR_EACH_EL_RF(el, rf) {
                     ^
external/opentimer/ot/timer/timer.cpp:1126:56: error: reference to local binding 'el' declared in enclosing function 'ot::Timer::_update_endpoints'
      std::sort(_endpoints[el][rf].begin(), _endpoints[el][rf].end());
                                                       ^
external/opentimer/ot/timer/timer.cpp:1105:18: note: 'el' declared here
  FOR_EACH_EL_RF(el, rf) {
                 ^
external/opentimer/ot/timer/timer.cpp:1126:60: error: reference to local binding 'rf' declared in enclosing function 'ot::Timer::_update_endpoints'
      std::sort(_endpoints[el][rf].begin(), _endpoints[el][rf].end());
                                                           ^
external/opentimer/ot/timer/timer.cpp:1105:22: note: 'rf' declared here
  FOR_EACH_EL_RF(el, rf) {
                     ^
external/opentimer/ot/timer/timer.cpp:1129:22: error: reference to local binding 'el' declared in enclosing function 'ot::Timer::_update_endpoints'
      if(!_endpoints[el][rf].empty()) {
                     ^
external/opentimer/ot/timer/timer.cpp:1105:18: note: 'el' declared here
  FOR_EACH_EL_RF(el, rf) {
                 ^

tsung-wei-huang, 2019-04-29:
Hi @renau , sorry for the late reply! I have updated this fix in the master branch. Please try it again :)

renau, 2019-04-29:
Hi,
Looks fine, but I noticed that you committed this file
https://github.com/OpenTimer/OpenTimer/blob/master/ot/config.hpp
It is good for me, before I had to do it myself.

Thanks for the patch,
Jose Renau
Professor, Computer Science & Engineering
W // http://masc.soe.ucsc.edu