OpenTimer/OpenTimer #30: report timing for a design with only clock in the SDC file

rohan-ganpati, 2019-04-30:
_When running the simple example provided in the Wiki, once the celllib and verilog are read if I only have create_clock -period 50 -name tau2015_clk [get_ports tau2015_clk] in the SDC file, **OpenTimer does not report timing:**_

```
____              _______
/ __ \___  ___ ___/  ___(_)_ _  ___ ____
/ /_/ /_ \/ -_)_ \/ / / /  ' \/ -_) __/
\____/ .__/\__/_//_/_/ /_/_/_/_/\__/_/       v2.0.0 (alpha)
   /_/
MIT License: type “license” to see more details.
For help, type “help”.
For bug reports, issues, and manual, please see:
<https://github.com/OpenTimer/OpenTimer>.
ot> read_celllib osu018_stdcells.lib
ot> read_verilog simple.v
ot> read_sdc simple.sdc
ot> report_timing -num_paths 5
I 63232 19-04-30 13:35:43 celllib.cpp:40] loading celllib “osu018_stdcells.lib”
I  5888 19-04-30 13:35:43 sdc.cpp:35] loading sdc “simple.sdc” ...
I  1792 19-04-30 13:35:43 verilog.cpp:18] loading netlist “simple.v”
I 63232 19-04-30 13:35:43 unit.cpp:256] use celllib time unit 1e-09 s
I 63232 19-04-30 13:35:43 unit.cpp:270] use celllib capacitance unit 1e-12 F
I 63232 19-04-30 13:35:43 unit.cpp:284] use celllib current unit 1e-06 A
I 63232 19-04-30 13:35:43 unit.cpp:298] use celllib voltage unit 1 V
I 63232 19-04-30 13:35:43 unit.cpp:312] use celllib resistance unit 1000 Ohm
I 63232 19-04-30 13:35:43 unit.cpp:326] use celllib power unit 1e-09 W
I 63232 19-04-30 13:35:43 celllib.cpp:73] added min celllib “osu018_stdcells” [cells:32]
I 63232 19-04-30 13:35:43 celllib.cpp:73] added max celllib “osu018_stdcells” [cells:32]
I 63232 19-04-30 13:35:43 verilog.cpp:25] added verilog module “simple” [gates:5]
I  5888 19-04-30 13:35:44 sdc.cpp:25] added 1 sdc commands
no critical path found

```
_**whereas OpenSTA does:**_

```
$../sta/sta 
OpenSTA Copyright (c) 2018, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
% read_verilog simple.v
Error: history.tcl, 330 invoked "return" outside of a proc.
1
% read_liberty osu018_stdcells.lib
1
% link_design simple
1
% create_clock -name tau2015_clk -period 1 {tau2015_clk}
% report_checks
Startpoint: f1 (falling edge-triggered flip-flop clocked by tau2015_clk)
Endpoint: f1 (falling edge-triggered flip-flop clocked by tau2015_clk)
Path Group: tau2015_clk
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.50    0.50   clock tau2015_clk (fall edge)
   0.00    0.50   clock network delay (ideal)
   0.00    0.50 v f1/CLK (DFFNEGX1)
   0.14    0.64 ^ f1/Q (DFFNEGX1)
   0.05    0.69 v u4/Y (NOR2X1)
   0.00    0.69 v f1/D (DFFNEGX1)
           0.69   data arrival time

   1.50    1.50   clock tau2015_clk (fall edge)
   0.00    1.50   clock network delay (ideal)
   0.00    1.50   clock reconvergence pessimism
           1.50 v f1/CLK (DFFNEGX1)
  -0.19    1.31   library setup time
           1.31   data required time
---------------------------------------------------------
           1.31   data required time
          -0.69   data arrival time
---------------------------------------------------------
           0.63   slack (MET)
```

tsung-wei-huang, 2019-04-30:
@rohan-ganpati , OpenSTA has different constraint definition from OpenTimer. You will need to specify the input delay/arrival-time and output delay in sdc file.

rohan-ganpati, 2019-05-01:
Yes, I understand that. But I am requesting if this feature could be added to OpenTimer?

(My bad, when I created the issue, there was a feature request option that I clicked but it does not appear on the issue, so I did not include the text again)

tsung-wei-huang, 2019-05-02:
I will add this to the todo list.

rohan-ganpati, 2019-05-02:
Thank you! :) Much appreciated.