OpenTimer/OpenTimer #36: Cmake -- configuring incomplete, errors occurred!

hey927, 2020-08-20:
"cmake ../" gave me the following result
-------------------------------------------------------------------------------------------
cmake ../
-- CMAKE_ROOT: /Applications/CMake.app/Contents/share/cmake-3.18
-- The C compiler identification is AppleClang 8.0.0.8000038
-- The CXX compiler identification is AppleClang 8.0.0.8000038
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
**-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped**
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
**-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped**
-- Detecting CXX compile features
-- Detecting CXX compile features - done
**CMake Error at CMakeLists.txt:23 (message):**
  

  OpenTimer currently supports the following compilers:

  - g++ v7.3 or above

  - clang++ v6.0 or above

**-- Configuring incomplete, errors occurred!**
See also "/Users/divyagupta/OpenTimer/build/CMakeFiles/CMakeOutput.log".
-------------------------------------------------------------------------------------------
I'm running on MacOS 10.11, I am not able to understand why the C & C++ compiler checks have been skipped.
XCode comes with gcc. I checked gcc version, Clang 8.0.0 :

gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.38)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

--------------------------------------------------------------------------------------------------------------
CMake error at CMakeLists.txt:23 is just what is shown in output, as below:

CMake Error at CMakeLists.txt:23 (message):
  

  OpenTimer currently supports the following compilers:

  - g++ v7.3 or above

  - clang++ v6.0 or above

--------------------------------------------------------------------------------------------------------------
How can I make the cmake run successfully? 

Thank you for your time!

tsung-wei-huang, 2020-08-20:
@hey927 You will need to use a g++ compiler.    Use the following configuration command to switch to G++:

```
~$ cmake ../ -DCMAKE_CXX_COMPILER=/path/to/your/g++
```

hey927, 2020-08-20:
I found the path to gcc using 'whereis gcc' command and used that to set in '~$ cmake ../ -DCMAKE_CXX_COMPILER=/path/to/your/g++'

still getting the same results about configuring incomplete.

[image]

hey927, 2020-08-20:
Also, tried changing paths to both C and C++ compiler as follows:
 '~$ cmake ../ -DCMAKE_CXX_COMPILER=/path/to/your/g++'
'~$ cmake ../ -DCMAKE_C_COMPILER=/path/to/your/gcc'

[image]

hey927, 2020-08-24:
My g++ is 4.2.1
g++ -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/c++/4.2.1

I can see OpenTimer supports g++ 7.3 or above.
As far as I could understand, Xcode comes with Command line tools (gcc and g++), which is being pointed to when I check the g++ version. Since, I had an earlier version of Xcode, I removed and it installed the latest Xcode (8.2.1 for my macOS- El Capitan), hoping to get latest g++ version with it. I'm still getting 👍🏽 
g++ -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1

Am i missing something?

Mobiappsolutionz, 2020-08-28:
Hello, 

 I am also facing same issue. 

-- CMAKE_ROOT: /Applications/CMake.app/Contents/share/cmake-3.18
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:23 (message):
  

  OpenTimer currently supports the following compilers:

  - g++ v7.3 or above

  - clang++ v6.0 or above

-- Configuring incomplete, errors occurred!
See also "/Users/gursimransinghsodhi/Documents/OpenTime/OpenTimer/build/CMakeFiles/CMakeOutput.log".

Please advise

huntingt, 2020-09-05:
If you want to fix the issue. It seems like the easiest way is to download gcc@8 with homebrew and use that to compile it -DCMAKE_C_COMPILER=<path to gcc-8> as given above (given when it's installed). If you try and use version 10 of gcc, it looks like a library changed so that there is now a conflict in namespace naming with the filename alias for experimental::filename and another filename namespace in the chrono library. Maybe gcc@9 works, I haven't looked into it.

Weijun-H, 2022-04-01:
This problem is because you use MacOS, the system will switch gcc/g++ to clang by default even you use ``` ~$ cmake ../ -DCMAKE_CXX_COMPILER=/path/to/your/g++ ```. You should determine the gcc version you used ``` gcc -dumpversion | cut -f1 -d```, then find the path of your g++ ```which g++-11```, for me it is g++ 11. Finally, use ``` ~$ cmake ../ -DCMAKE_CXX_COMPILER=G++_PATH ```