OpenTimer/OpenTimer #38: what does _pin.find(name)== _pins.end() failed  mean ? how did this failed arised ,and how can it be resolved

fxh-1234, 2020-09-21:
**Title**
Use square bracket to highlight the request. Ex. [API request]

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the API you'd like**
A clear and concise description of the API spec you need for this feature.

buraktoker, 2021-08-24:
Same issue any solution??

gxydbc, 2024-09-28:
> **Title** Use square bracket to highlight the request. Ex. [API request]
>
> **Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is. [...]

I am also facing a similar issue. Have you figured out the solution for this issue?

ThunderCarver, 2024-12-01:
I suppose that the _pin.find(name)== _pins.end() wants to find the pin refered as "name". If the result is true, there is no pin called "name". 
However you said a failure rasied, could you please provide more details about the failure?

gxydbc, 2024-12-02:
> I suppose that the _pin.find(name)== _pins.end() wants to find the pin refered as "name". If the result is true, there is no pin called "name". However you said a failure rasied, could you please provide more details about the failure?

I rechecked and found that my Verilog design involves hierarchical designs with multiple modules, including bus definitions. I discovered in the Verilog documentation on the wiki that the issues will be addressed in a future update. How can I obtain the new version?

ThunderCarver, 2024-12-05:
The newest version of Opentimer is current version. I am not sure the version of Verilog Parser. Based on your description, it looks like the problem is related with Verilog Parser. Perhaps you can inspect the verilog parser used in Opentimer and see if there is new version released.

gxydbc, 2024-12-05:
I tried the latest version， but i still met the same problems. It seems that the Verilog parser still does not support complex Verilog. If modifications are needed, would it be very complicated?

> The newest version of Opentimer is current version. I am not sure the version of Verilog Parser. Based on your description, it looks like the problem is related with Verilog Parser. Perhaps you can inspect the verilog parser used in Opentimer and see if there is new version released.

ThunderCarver, 2024-12-05:
> I tried the latest version， but i still met the same problems. It seems that the Verilog parser still does not support complex Verilog. If modifications are needed, would it be very complicated?

Sorry, I am not a specialist for verilog parser. I am just studying on OpenTimer recently. Perhaps you can change the name to see if it works. I guess the hierarchical structure works on verilog module's name as well.