OpenTimer/OpenTimer #39: Issue compiling on mac

rsetaluri, 2020-10-07:
Hi, I am trying to compile on mac (macOS Catalina 10.15.6). Having trouble building with both clang and gcc.

## Clang

First issue was I had to change L18 of `CMakeLists.txt` to be  `"AppleClang"` instead of `"Clang"` to get `cmake` to work. When I run `make` I get:

```
/usr/local/Cellar/cmake/3.14.0/bin/cmake -S/home/OpenTimer -B/home/OpenTimer/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/Cellar/cmake/3.14.0/bin/cmake -E cmake_progress_start /home/OpenTimer/build/CMakeFiles /home/OpenTimer/build/CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/OpenTimer.dir/build.make CMakeFiles/OpenTimer.dir/depend
cd /home/OpenTimer/build && /usr/local/Cellar/cmake/3.14.0/bin/cmake -E cmake_depends "Unix Makefiles" /home/OpenTimer /home/OpenTimer /home/OpenTimer/build /home/OpenTimer/build /home/OpenTimer/build/CMakeFiles/OpenTimer.dir/DependInfo.cmake --color=
Scanning dependencies of target OpenTimer
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/OpenTimer.dir/build.make CMakeFiles/OpenTimer.dir/build
[  1%] Building CXX object CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -I/home/OpenTimer  -Wall -O2 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk   -std=c++1z -o CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o -c /home/OpenTimer/ot/unit/unit.cpp
In file included from /home/OpenTimer/ot/unit/unit.cpp:1:
In file included from /home/OpenTimer/ot/unit/unit.hpp:4:
In file included from /home/OpenTimer/ot/headerdef.hpp:52:
/home/OpenTimer/ot/patch/clang_variant.hpp:38:10: fatal error: 'bits/enable_special_members.h' file not found
#include <bits/enable_special_members.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o] Error 1
make[1]: *** [CMakeFiles/OpenTimer.dir/all] Error 2
make: *** [all] Error 2
```

Clang version is

```
Apple clang version 12.0.0 (clang-1200.0.32.2)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

## GCC

If instead use g++ (`/usr/local/bin/g++-10`) `cmake` is successful, but `make` gives

```
/usr/local/Cellar/cmake/3.14.0/bin/cmake -S/home/OpenTimer -B/home/OpenTimer/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/Cellar/cmake/3.14.0/bin/cmake -E cmake_progress_start /home/OpenTimer/build/CMakeFiles /home/OpenTimer/build/CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/OpenTimer.dir/build.make CMakeFiles/OpenTimer.dir/depend
cd /home/OpenTimer/build && /usr/local/Cellar/cmake/3.14.0/bin/cmake -E cmake_depends "Unix Makefiles" /home/OpenTimer /home/OpenTimer /home/OpenTimer/build /home/OpenTimer/build /home/OpenTimer/build/CMakeFiles/OpenTimer.dir/DependInfo.cmake --color=
Scanning dependencies of target OpenTimer
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/OpenTimer.dir/build.make CMakeFiles/OpenTimer.dir/build
[  1%] Building CXX object CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o
/usr/local/bin/g++-10   -I/home/OpenTimer  -Wall -O2 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk   -std=c++17 -o CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o -c /home/OpenTimer/ot/unit/unit.cpp
In file included from /home/OpenTimer/ot/unit/unit.hpp:4,
                 from /home/OpenTimer/ot/unit/unit.cpp:1:
/home/OpenTimer/ot/headerdef.hpp:67:50: error: 'namespace std::filesystem = std::experimental::std::experimental::filesystem;' conflicts with a previous declaration
   67 |   namespace filesystem = experimental::filesystem;
      |                                                  ^
In file included from /usr/local/Cellar/gcc/10.2.0/include/c++/10.2.0/thread:43,
                 from /home/OpenTimer/ot/headerdef.hpp:7,
                 from /home/OpenTimer/ot/unit/unit.hpp:4,
                 from /home/OpenTimer/ot/unit/unit.cpp:1:
/usr/local/Cellar/gcc/10.2.0/include/c++/10.2.0/chrono:54:13: note: previous declaration 'namespace std::filesystem { }'
   54 |   namespace filesystem { struct __file_clock; };
      |             ^~~~~~~~~~
make[2]: *** [CMakeFiles/OpenTimer.dir/ot/unit/unit.cpp.o] Error 1
make[1]: *** [CMakeFiles/OpenTimer.dir/all] Error 2
make: *** [all] Error 2
```

GCC version is

```
g++-10 (Homebrew GCC 10.2.0) 10.2.0
```

edwintorok, 2020-11-23:
Same issue on Fedora 33 with g++ (GCC) 10.2.1 20201016 (Red Hat 10.2.1-6)

hughperkins, 2022-03-07:
Thank-you for hte heads-up about changing line 18.

hughperkins, 2022-03-07:
and yes I get
```
/Users/hp/git/OpenTimer/ot/patch/clang_variant.hpp:38:10: fatal error: 'bits/enable_special_members.h'
      file not found
#include <bits/enable_special_members.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

hughperkins, 2022-03-07:
I guess off to Parallels for now .... :/

hughperkins, 2022-03-07:
so, actually, it turns out that stuff compiles a bit more if one opens headerdef.hpp, and changes lines 51-55, to comment out clang_variant:

```
//#if defined(__clang__)
//  #include <ot/patch/clang_variant.hpp>
//#else
  #include <variant>
//#endif
```

No idea how far this will get. It's at 50% as I type this...

hughperkins, 2022-03-07:
Well, got to 82%, then linker errors

```
[ 82%] Linking CXX executable ../example/simple/simple
/usr/local/Cellar/cmake/3.22.1/bin/cmake -E cmake_link_script CMakeFiles/simple.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++  -Wall -O2 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/simple.dir/example/simple/simple.cpp.o -o ../example/simple/simple  ../lib/libOpenTimer.a -lstdc++fs 
1 warning generated.
ld: library not found for -lstdc++fs
ld: library not found for -lstdc++fs
ld: library not found for -lstdc++fs
clangclangclang: : : errorerrorerror: : : linker command failed with exit code 1 (use -v to see invocation)linker command failed with exit code 1 (use -v to see invocation)linker command failed with exit code 1 (use -v to see invocation)
```

hughperkins, 2022-03-07:
so, I opened CMakeLists.txt, and commented out line line 169, and build finished ok...
```
# list(APPEND OT_LINK_FLAGS stdc++fs)
```

```
[ 98%] Linking CXX executable ../example/sizer/sizer
/usr/local/Cellar/cmake/3.22.1/bin/cmake -E cmake_link_script CMakeFiles/sizer.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++  -Wall -O2 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/sizer.dir/example/sizer/sizer.cpp.o -o ../example/sizer/sizer  ../lib/libOpenTimer.a 
[ 98%] Built target sizer
1 warning generated.
[100%] Linking CXX executable ../unittest/path
/usr/local/Cellar/cmake/3.22.1/bin/cmake -E cmake_link_script CMakeFiles/path.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++  -Wall -O2 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/path.dir/unittest/path.cpp.o -o ../unittest/path  ../lib/libOpenTimer.a 
[100%] Built target path
/usr/local/Cellar/cmake/3.22.1/bin/cmake -E cmake_progress_start /Users/hp/git/OpenTimer/build/CMakeFiles 0
(cosu38) cosu38 ~/git/OpenTimer/build (master|…10△2)$ 
```

hughperkins, 2022-03-07:
seg faults if I try running it though...
```
(cosu38) cosu38 ~/git/OpenTimer/build (master|…10△2)$ ot-shell
  ____              _______              
 / __ \___  ___ ___/_  __(_)_ _  ___ ____
/ /_/ / _ \/ -_) _ \/ / / /  ' \/ -_) __/
\____/ .__/\__/_//_/_/ /_/_/_/_/\__/_/       v2.1.0
    /_/                                     
MIT License: type "license" to see more details.
For help, type "help".
For bug reports, issues, and manual, please see:
<https://github.com/OpenTimer/OpenTimer>.
ot> cd example/simple
No such file or directory: example/simple
ot> ls
CMakeCache.txt		CTestTestfile.cmake	Makefile		cmake_install.cmake
CMakeFiles		DartConfiguration.tcl	Testing			install_manifest.txt
ot> cd ../ex
example    
ot> cd ../example/si
simple    sizer     
ot> cd ../example/sim
simple    
ot> cd ../example/simple/
ot> read_celllib osu018_stdcells.lib
Segmentation fault: 11
```

gzz2000, 2022-04-12:
I am able to compile on macos 11.2.3 (M1 chip), with **g++ 11** from homebrew, after doing the following changes: https://github.com/gzz2000/opentimer-patches/commit/f2189b1da01dc0a7b14af9d4da9a257863ad53aa