OpenTimer/OpenTimer #41: Compilation failed with higher version of g++

enzoleo, 2021-03-03:
With `g++10.2` the compilation will fail due to the following lines. These lines appear in multiple files and induce duplicate declaration.
```c++
namespace std {
  namespace filesystem = experimental::filesystem;
};
```
I believe with higher version of `g++` the namespace `std::filesystem` has ready been declared in the standard header `<filesystem>`. Replace all `std::experimental::filesystem` with `std::filesystem` and `<experimental/filesystem>` with `<filesystem>` the problem can be resolved (but this workaround does not consider compatibility).

tsung-wei-huang, 2021-03-03:
Yes, I will change it later on - for now, feel free to modify that namespace to get rid of the compilation error.

mooredan, 2021-06-09:
Me too. I'm on a Pop!_OS 20.10 system with c++ (Ubuntu 10.2.0-13ubuntu1) 10.2.0

These sed commands got this compiling:

```
sed -i -e 's:experimental/filesystem:filesystem:' ./ot/parser-spef/parser-spef.hpp ./ot/shell/prompt.hpp ./ot/headerdef.hpp ./ot/utility/os.hpp ./ot/utility/tokenizer.hpp

sed -i -e 's/std::experimental::filesystem/std::filesystem/' ./ot/parser-spef/parser-spef.hpp

sed -i -e 's/experimental::filesystem/std::filesystem/' ./ot/shell/prompt.hpp ./ot/utility/os.hpp ./ot/utility/tokenizer.hpp

```
I can provide a pull request, but I suspect this isn't the correct way to address this as it might cause  backward compatibility issue.

PR #45 (2021-07-13): Fix build with new C++ compilers/libraries
The first commit is a missing include and hopefully uncontroversial.

The second commit fixes the build on compilers where `std::filesystem` already exists, by aliasing `ot::filesystem` to `std::experimental::filesystem` rather than attempting to overwrite `std::filesystem` which produces an error in this case.

I think this should fix #41, too.
closed 2021-07-13 by tsung-wei-huang