OpenTimer/OpenTimer #55: After installing cmake it is showing no cxx compiler found. I have installed cmake and checked by using the command which cmake but after running the code cmake ../ error is coming. Any solutions?

SAPTARSHI5, 2022-02-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the bug.

oliverbunting, 2022-02-13:
Try `which gc++`. If it's not available, install the Ubuntu package build-essentials

SAPTARSHI5, 2022-02-28:
Thanks for the reply currently I am facing this issue whose screenshot i have attached here
[image]

SAPTARSHI5, 2022-02-28:
Could you please help me to resolve this? I have been constantly trying to run the tool.

SAPTARSHI5, 2022-02-28:
Hi tsung-wei-huang,
Thanks for your reply and helping me.
I have checked the gcc version and its above it 8.2. I have attached a screenshot for your reference
[image]

hughperkins, 2022-03-07:
built ok for me on ubuntu 20.04 out of the box, I have same version as you, oddly:
```
ubuntu@ubuntu:/media/psf/OpenTimer/example/simple$ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

ubuntu@ubuntu:/media/psf/OpenTimer/example/simple$ g++ --version
g++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

hughperkins, 2022-03-07:
I'm building `master`, cloned ysterday.

Here is my CMakeCache.txt, in case useful https://gist.github.com/hughperkins/42b988a33e38b09952bcf2da4fa0ed5e

hughperkins, 2022-03-07:
[image]