OpenTimer/OpenTimer #56: Unable to install OpenTimer

albyne1602, 2022-02-19:
[image]
[image]
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the bug.

qrqiuren, 2022-04-02:
Could you please run the following command and see the result?

```
g++ --version
/usr/bin/c++ --version
```

sijaykumar, 2022-06-15:
Hi Guys, I am also getting two error.

In my ubuntu g++ compiler g++ (Version 8.5.0) and /usr/bin/c++(Version 8.5.0) are installed
 

#############################################################
n file included from /home/sijay/OpenTimer/unittest/utility.cpp:3:
/home/sijay/OpenTimer/unittest/doctest.h:4008:57: error: array bound is not an integer constant before ‘]’ token
         static char             altStackMem[4 * SIGSTKSZ];
                                                         ^
/home/sijay/OpenTimer/unittest/doctest.h: In constructor ‘doctest::{anonymous}::FatalConditionHandler::FatalConditionHandler()’:
/home/sijay/OpenTimer/unittest/doctest.h:4027:33: error: ‘altStackMem’ was not declared in this scope
             sigStack.ss_sp    = altStackMem;
                                 ^~~~~~~~~~~
/home/sijay/OpenTimer/unittest/doctest.h: At global scope:
/home/sijay/OpenTimer/unittest/doctest.h:4056:57: error: ‘char doctest::{anonymous}::FatalConditionHandler::altStackMem []’ is not a static data member of ‘struct doctest::{anonymous}::FatalConditionHandler’
     char             FatalConditionHandler::altStackMem[]                              = {};
                                                         ^
/home/sijay/OpenTimer/unittest/doctest.h:4056:22: warning: ‘doctest::{anonymous}::FatalConditionHandler::altStackMem’ defined but not used [-Wunused-variable]
     char             FatalConditionHandler::altStackMem[]                              = {};
                      ^~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/utility.dir/build.make:66: CMakeFiles/utility.dir/utility.cpp.o] Error 1
make[2]: Leaving directory '/home/sijay/OpenTimer/unittest'
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/utility.dir/all] Error 2
make[1]: Leaving directory '/home/sijay/OpenTimer/unittest'
make: *** [Makefile:144: all] Error 2
############################################################