OpenTimer/OpenTimer #58: Question: how to generate the SDC file?

hughperkins, 2022-03-07:
Hi, OpenTimer looks interesting :)

In the 'simple example', there is a verilog netlist file, which I suppose I can get from my behavioral verilog file using yosys; and there is an sdc file. How do I get the sdc file, using opensource tools?

qrqiuren, 2022-04-02:
It's the timing constraint file that you must define yourself. You can either search for Synopsys' manual or read the book [Constraining Designs for Synthesis and Timing Analysis](https://laesferadigital.com/reading/constraining-designs-for-synthesis-and-timing-analysis/).

hughperkins, 2022-04-02:
Hmmm, I wrote my own script in the end https://github.com/hughperkins/VeriGPU/blob/main/verigpu/timing.py Seemed easier than figuring out how to write an sdc file :) Not sure if I'm missing anything from my script that opentimer handles, or handles better? My script estimates both maximum propagation delay, as a ratio of nand gate units; and area, also as a ratio of nand gate units.

qrqiuren, 2022-04-02:
I think your understanding is wrong.

The SDC file is the *constraint* of timing. For example, I want my circuit to run in 100MHz, then I can define a clock in 100MHz and let the compiler (`yosys` or some other things) to optimize the circuit to run in 100MHz. With the resulting netlist and the SDC, the STA tools (like OpenTimer) can check if your circuit meets such requirements.

If you cannot provide an SDC file to the STA tool, it could not perform such analysis because it cannot infer your timing paths.

Just let me show a simple example of defining the clock `clk` in 100MHz. You can pick it up and try to expand yourself with the manuals and STA books. By the way, the SDC format is just a subset of the Tcl language.

```tcl
create_clock -period 10 [get_ports {clk} ] -name clk
set_clock_uncertainty -setup 0.1 [get_clocks *]
set_clock_uncertainty -hold 0.1 [get_clocks * ]
set_clock_latency -source -fall -early 0.1 [get_clocks *]
set_clock_latency -source -fall -late 0.1 [get_clocks *]
set_clock_latency 0.5 [get_clocks *]
```

qrqiuren, 2022-04-02:
If you provide such SDC file to `yosys`, you may export the resulting netlist and the resulting SDC (not the original hand-written SDC). You can let OpenTimer to read them.