OpenTimer/OpenTimer #68: error processing verilog in ot/verilog/verilog.cpp

Yang-Xijie, 2022-07-27:
## Describe the bug

`ot/verilog/verilog.cpp` cannot process my verilog demo file...

## To Reproduce

This is my verilog file generated by yosys, which is an inverter:

**./output/inverter.v**

```verilog
/* Generated by Yosys 0.9 (git sha1 1979e0b) */

(* top =  1  *)
(* src = "inverter.v:1" *)
module inverter(in, out);
  (* src = "inverter.v:2" *)
  wire _0_;
  (* src = "inverter.v:3" *)
  wire _1_;
  (* src = "inverter.v:2" *)
  input in;
  (* src = "inverter.v:3" *)
  output out;
  sky130_fd_sc_hd__clkinv_1 _2_ (
    .A(_0_),
    .Y(_1_)
  );
  assign _0_ = in;
  assign out = _1_;
endmodule
```

Then I use the following commands to generate the time report:

**ot_inverter.txt**

```
read_celllib ./lib/sky130_fd_sc_hd__tt_025C_1v80.lib
read_verilog ./output/inverter.v
read_sdc inverter.sdc
report_timing
```

```
$ /home/yxj/VerilogOJ/OpenTimer/bin/ot-shell -i ot_inverter.txt
I 34560 22-07-27 17:38:35 celllib.cpp:34] loading celllib "./lib/sky130_fd_sc_hd__tt_025C_1v80.lib"
I 38656 22-07-27 17:38:35 verilog.cpp:14] loading netlist "./output/inverter.v"
I 42752 22-07-27 17:38:35 sdc.cpp:35] loading sdc "inverter.sdc" ...
F 38656 22-07-27 17:38:35 verilog.cpp:160] missing ; in instance declaration
Aborted (core dumped)
```

`F 38656 22-07-27 17:38:35 verilog.cpp:160] missing ; in instance declaration` showed error in Verilog parsing...

---

I then removed all the comments:

**./output/inverter.v**

```verilog
module inverter(in, out);
  wire _0_;
  wire _1_;
  input in;
  output out;
  sky130_fd_sc_hd__clkinv_1 _2_ (
    .A(_0_),
    .Y(_1_)
  );
  assign _0_ = in;
  assign out = _1_;
endmodule
```

```
$ /home/yxj/VerilogOJ/OpenTimer/bin/ot-shell -i ot_inverter.txt
I 34560 22-07-27 17:48:10 verilog.cpp:14] loading netlist "./output/inverter.v"
I 42752 22-07-27 17:48:10 celllib.cpp:34] loading celllib "./lib/sky130_fd_sc_hd__tt_025C_1v80.lib"
I 38656 22-07-27 17:48:10 sdc.cpp:35] loading sdc "inverter.sdc" ...
F 34560 22-07-27 17:48:10 verilog.cpp:156] syntax error in gate pin-net mapping
Aborted (core dumped)
```

I wonder what `F 34560 22-07-27 17:48:10 verilog.cpp:156] syntax error in gate pin-net mapping` means and how to fix this `syntax error`...

## Others

I am just a novice in the world of Verilog... Thank you for any hint!

chaitalisathe, 2022-11-28:
@Yang-Xijie I am also facing a similar issue. Have you figured out the solution for this issue?

Yang-Xijie, 2022-11-28:
@chaitalisathe Nope. We used [OpenSTA](https://github.com/The-OpenROAD-Project/OpenSTA) ([doc](https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/doc/OpenSTA.pdf)) instead.

Max-astro, 2023-01-30:
> @Yang-Xijie I am also facing a similar issue. Have you figured out the solution for this issue?

OpenTimer only support gate level Verilog syntex, it can not parse `assign` statement.